michalmonday/CSV-Parser-for-Arduino #1: byte format

hmronline, 2020-07-12:
Could it be possible to add byte or uint8_t to the [supported value type list](https://github.com/michalmonday/CSV-Parser-for-Arduino#specifying-value-types)?

Thanks,
Hernán.-

michalmonday, 2020-07-12:
Hi, I'll add the unsigned specifiers for 8/16/32 bit integers soon.

Btw it's nice to see that someone is using the library.

michalmonday, 2020-07-14:
It's possible now to use unsigned types (from version 0.1.2). I couldn't think of any characters that would be intuitive format specifiers for 4 different existing types ("c", "d", "L", "x"), so I thought that the existing ones can be preceeded with "u" instead, like:
"uc" - uint8_t
"ud" - uint16_t
"uL" - uint32_t
"ux" - uint32_t (using hexadecimal as input)

```cpp
  const char * csv_str = "bytes,words,dwords,dwords_from_hex\n"
                         "255,65535,4294967295,FFFFFFFF\n"
                         "254,65534,4294967294,FFFFFFFE\n";

  CSV_Parser cp(/*format*/ "ucuduLux");
```

I added full example here:
https://github.com/michalmonday/CSV-Parser-for-Arduino/blob/master/examples/unsigned_values/unsigned_values.ino

hmronline, 2020-07-14:
That's great, it seems it's working as expected. Thank you!

closed 2020-07-14 by hmronline