michalmonday/CSV-Parser-for-Arduino #2: Reading from SDCard example

hmronline, 2020-07-13:
I'm having issues to read a csv file from sdcard in chunks; it randomly fails to parse file, using the following code:

```
    objFile = SD.open("file.csv", "r");
    if (!objFile)
    {
        Serial.print(F("ERROR, unable to open csv file"));
        Serial.println(F(""));
        while (1)
            ;
    }

    */ file contents:
    key,value\n
    102,750\n
    101,750\n
    103,750\n
    104,750\n
    */

    CSV_Parser csvParser(/*format*/ "dd", /*has_header*/ true, /*delimiter*/ ',');

    char buffer[600];
    while (objFile.position() < objFile.size())
    {
        objFile.readBytes(buffer, sizeof(buffer));
        csvParser << buffer;
    }

    csvParser.parseLeftover();
    objFile.close();

```

I have verified that everything works fine if csv file contents is provided all at once as first parameter to csvParser.

Could you please tell the appropriate method to read csv chunks from sdcard ?

This could possible be included in the [provided example](https://github.com/michalmonday/CSV-Parser-for-Arduino/blob/master/examples/supplying_csv_by_incomplete_parts/supplying_csv_by_incomplete_parts.ino) to help others that may experience the same challenge.

Thanks in advance!
Hernán.-

michalmonday, 2020-07-13:
Thank you for reporting this.
There are 2 problems:
- library bug when supplying csv by chunks, fixed with these commits: [1](https://github.com/michalmonday/CSV-Parser-for-Arduino/commit/f9884a26149d5cc214b124dcbefd2cfe6baa11dd), [2](https://github.com/michalmonday/CSV-Parser-for-Arduino/commit/86f6d6af50a8c575a59383e393c41df0443b02a9), [3](https://github.com/michalmonday/CSV-Parser-for-Arduino/commit/a228f981f281bcf2d2b4f3d405b60f93a046e81f)  
- buffer in the example you posted is not terminated by 0, which can be fixed by defining it as "{0}" at declaration like:
```cpp
char buffer[600] = {0}; // it will fill it all with 0's
```
or alternatively, you could use the return of readBytes to terminate the string at the right place directly like:
```cpp
int i = objFile.readBytes(buffer, sizeof(buffer));
buffer[i] = 0;
```

I think it could be good idea to supply the file character by character, this way there's not need for using a buffer at all, I added example like this:
https://github.com/michalmonday/CSV-Parser-for-Arduino/blob/master/examples/reading_from_sd_card/reading_from_sd_card.ino

hmronline, 2020-07-13:
Hi Michal, thank you for your kick response!

I've tested the new version and I can confirm it now works properly following the new example code.

closed 2020-07-13 by hmronline