michalmonday/CSV-Parser-for-Arduino #3: Header fields

hmronline, 2020-07-26:
Hi Michal,

I was having an exception when trying to retrieve values, with a similar code to this:
```
char  **strings = (char**)cp["my_strings"];
```
I've got exception 28 (LoadProhibitedCause) on that line of code and I've spent almost a day troubleshooting this.

Finally found that the exception was caused because of having spaces on header fields, like this:
```
my_strings, my_floats
```
Note the space before "my_floats" header field.

I'm not sure if something can be done on library to prevent this, but I suggest you to at least put some note there to prevent others doing something stupid like putting an space between fields like I did.

Thank you for this great library!

michalmonday, 2020-07-27:
Hi, from the 0.1.3 release the leading and trailing spaces in header fields are ignored.

Example: 
```cpp
char * csv_str = "  test a  ,  test b  \n" 
     "1,2\n"
     "3,4\n";
	 
CSV_Parser cp(csv_str, "ss");
char  **a = (char**)cp["test a"]; // "test a" is used instead of "  test a  "
char  **b = (char**)cp["test b"];
```
Btw if the specified key is not found then cp[key] should return 0, allowing to implement check like:
```cpp
char  **strings = (char**)cp["my_strings"];
if (strings) {
  // get values
} else {
  Serial.println("'my_strings' key was not found, values can't be parsed.");
}
```

If the LoadProhibitedCause exception was raised even when such check was implemented, then handling of "[]" operator in this library may be bugged, because it shouldn't raise exception itself, it should just return 0. Attempting to read from 0 address would raise exception, I hope that was the case.

hmronline, 2020-07-31:
Works fine, Thank you!

closed 2020-07-31 by hmronline