michalmonday/CSV-Parser-for-Arduino #7: Two requests

wbadry, 2021-05-19:
Hello,
The parser looks great and in a very structured coding.

My two requests, if possible:

- To be able to use `Serial` , `Serial1` or `Serial2` with `cp.print()`
- To be able to configure the SD pins when using `cp.readSDfile()`. This will be very handy when it comes to ESP32, which uses HSPI and VSPI that are native SPI than SPI, which is software in its case. The SD library allows using custom pins for SPI for this.

Thank you so much.

michalmonday, 2021-05-19:
Hello, thank you for suggestions but I think that both are somewhat implemented already.
cp.print() method has a "Stream" parameter of which default value is "Serial", so if anyone would like to provide different Serial object, it could be done like:

```cpp
cp.print(Serial2);
```
Because it's declared like:
[image]

Regarding the configurable pins of SD card, if you take a look at the [reading_from_SD_card example](https://github.com/michalmonday/CSV-Parser-for-Arduino/blob/master/examples/reading_from_sd_card/reading_from_sd_card.ino), the CSV-Parser library itself does not initialize the SD card, it has to be initialized/configured externally (by calling "SD.begin(chip_select_pin)"). CSV-Parser only calls "SD.open()" (inside the "cp.readSDfile()" method) and then operates on "File" object (calling File.available, File.read and File.close methods).

So as long as the [SD library](https://www.arduino.cc/en/reference/SD) allows to configure pins, the CSV-Parser will not interfere with it.

From what I see, [customisation of pins for ESP32](https://www.lucadentella.it/en/2021/01/13/esp32-arduino-sd-card-con-pin-custom/) takes place within the "SD.begin" method, so I guess it should work fine.

wbadry, 2021-05-19:
Thank you so much, @michalmonday, for your prompt response.  This is fantastic indeed.

closed 2021-05-19 by wbadry