michalmonday/CSV-Parser-for-Arduino #8: readSDfile path

pedro-nonfree, 2021-06-06:
hi @michalmonday , congratulations for such an amazing piece of software :)

we are using the *reading from sd card* example

in our case, to correctly read the file we had to place a `/` before the filename

```diff
-  cp.readSDfile("file.csv"); // this wouldn't work if SD.begin wasn't called before
+  cp.readSDfile("/file.csv"); // this wouldn't work if SD.begin wasn't called before
```

the upstream documentation was lastly reviewed on ~2018 so maybe something changed :shrug: https://www.arduino.cc/en/Tutorial/DumpFile

it might be useful too if there is some kind of checking such as:

```
file_read_ok =  cp.readSDfile("file.csv"); // this wouldn't work if SD.begin wasn't called before
if( ! file_read_ok ) {
  Serial.println("cannot read file from sdcard, check the path is correct");
  // don't know if `exit(1);` would work works
}
```

michalmonday, 2021-06-06:
Hi, thank you for letting me know, I edited the filename just in case. It's a bit weird though because the [reference page of SD library](https://www.arduino.cc/en/reference/SD) states:
> Because the working directory is always the root of the SD card, a name refers to the same file whether or not it includes a leading slash (e.g. "/file.txt" is equivalent to "file.txt").

Btw by looking at the [SD library history](https://github.com/arduino-libraries/SD/commits/master/src), it doesn't seem that there were any relevant changes since 2018. There is [one from 2017](https://github.com/arduino-libraries/SD/commit/f4f3b0c62386822e690999d25e87eadef05cf8e3#diff-25a6634263c1b1f6fc4697a04e2b9904ea4b042a89af59dc93ec1f5d44848a26) which at first glance could be related but I have no idea if it has something to do with this issue.

Also, thank you for the error checking suggestion, I just added it in [this commit](https://github.com/michalmonday/CSV-Parser-for-Arduino/commit/5e542ba561bd3faf8cc2198913b7bdcbd952f79d).

pedro-nonfree, 2021-06-08:
I was helping a friend, the answer is with @smjacques (ping!)

smjacques, 2021-06-10:
Hi! The SD lib version is 1.2.4 and the Arduino IDE is the 1.8.13.