michalmonday/CSV-Parser-for-Arduino #9: SdFat version?

hicksan, 2021-09-10:
This looks like a really fantastic library and I would love to be able to use it.

I am building a project on an ESP8266 Wemos D1 board and therefore have to use the ESP8266 version of SdFat.

Can this library be made to work with SdFat instead of SD.h? I have tried but not succeeded getting anything to compile.

michalmonday, 2021-09-11:
Hello, I'm using Arduino 1.8.15 and Esp8266 boards for Arduino package (2.5.0). I tried to compile [SD card example](https://github.com/michalmonday/CSV-Parser-for-Arduino/blob/master/examples/reading_from_sd_card/reading_from_sd_card.ino) for "Generic esp8266 board" and "LOLIN(WEMOS) D1 mini Pro" and in both cases it compiled successfully.

Are you using Platformio by any chance? If yes, then possibly the suggestion from [this issue](https://github.com/michalmonday/CSV-Parser-for-Arduino/issues/5) will be helpful.

I never used SdFat library on its own, it seems low level and I'm not sure how to integrate it. If you manage to read a file byte by byte (or in chunks), then you could just supply it to CSV_Parser object as shown in [this example](https://github.com/michalmonday/CSV-Parser-for-Arduino/blob/master/examples/supplying_csv_by_incomplete_parts/supplying_csv_by_incomplete_parts.ino).

hicksan, 2021-09-11:
The following code works with SdFat:

Looking at CSV-Parser.cpp, line 4 :
#include <SD.h>
I have replaced this with:
#include <SdFat.h>
using namespace sdfat;
SdFat sd;

Then line 111:
File csv_file = SD.open(f_name);
if (!csv_file) {
...
becomes:
File csv_file;
if (!csv_file.open(f_name, FILE_READ)) {
...

This works with SdFat.

Yes SD.h does work with ESP8266, but not with all other libraries and I was unable to get it working with my particular configuration of modules where I finally managed to get SdFat working. My understanding is that SdFat is more recent, more powerful and better supported - but I may be completely wrong. Subject to the changes above it works fine and my CSV files are parsing beautifully. Job done.

Thank you so much for a fantastic library that has saved me a whole pile of work. And thank you also for responding to issue threads - I know it must be a huge burden but it really helps users getting started and want you to know how much it is appreciated.

closed 2021-09-11 by hicksan