michalmonday/CSV-Parser-for-Arduino #10: Unable to compile on espressif8266 board

Zirafarafa, 2021-09-11:
CSV-Parser requires SD (and SPI), but I get a compilation error on an espressif8266 board.

```
`SD/utility/Sd2PinMap.h:371:2: error: #error Architecture or board not supported.`
```

`Sd2PinMap.h` seems to only support arm:

```
#if defined(__arm__) // Arduino Due Board follows
```

Is there any way to use CSV-Parser without importing the SD dependencies?

michalmonday, 2021-09-11:
Hi, it seems that it happened to people in the past, someone suggested a solution in this issue:
https://github.com/michalmonday/CSV-Parser-for-Arduino/issues/5

However, I just added an option to disable it by using the following `#define` before the inclusion of the library:

```cpp
#define CSV_PARSER_DONT_IMPORT_SD
#include <CSV_Parser.h>
```

Note that it must be positioned above the import, otherwise it won't take effect.
Btw, having this `#define` disables the `CSV_Parser::readSDfile` method, that's because this method is relying on the SD library.
This change is introduced in 0.2.1 version, it may take some time until this version is available in library manager. You could manually download it from this repository though.

It would be great if you confirmed whether it solves the problem or not (because I didn't use Platformio to test, but I think this should solve the problem).

Zirafarafa, 2021-09-20:
Hi - just a note that I have not had a chance to try the fix yet, as my project has been put aside for a while