michalmonday/CSV-Parser-for-Arduino #16: Compile fails on RP2040

JoeEarlam, 2022-10-27:
Hi there! Using the basic_usage example, compilation fails with an RP2040 chip as the target (even with CSV_PARSER_DONT_IMPORT_SD defined).

```
/home/jearlam/Arduino/libraries/CSV-Parser-for-Arduino/examples/basic_usage/basic_usage.ino: In function 'void setup()':
/home/jearlam/Arduino/libraries/CSV-Parser-for-Arduino/examples/basic_usage/basic_usage.ino:26:20: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   26 |   char * csv_str = "my_strings,my_numbers\n"
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~~
   27 |                    "hello,5\n"
      |                    ~~~~~~~~~~~
   28 |                    "world,10\n";
      |                    ~~~~~~~~~~~~
In file included from /home/jearlam/.arduino15/packages/rp2040/hardware/rp2040/2.6.2/libraries/SDFS/src/SDFS.h:35,
                 from /home/jearlam/.arduino15/packages/rp2040/hardware/rp2040/2.6.2/libraries/SD/src/SD.h:24,
                 from /home/jearlam/Arduino/libraries/CSV-Parser-for-Arduino/CSV_Parser.cpp:6:
/home/jearlam/Arduino/libraries/SdFat/src/SdFat.h:452:2: warning: #warning File not defined because __has_include(FS.h) [-Wcpp]
  452 | #warning File not defined because __has_include(FS.h)
      |  ^~~~~~~
In file included from /home/jearlam/.arduino15/packages/rp2040/hardware/rp2040/2.6.2/libraries/SD/src/SD.h:24,
                 from /home/jearlam/Arduino/libraries/CSV-Parser-for-Arduino/CSV_Parser.cpp:6:
/home/jearlam/.arduino15/packages/rp2040/hardware/rp2040/2.6.2/libraries/SDFS/src/SDFS.h: In member function 'virtual int sdfs::SDFSFileImpl::availableForWrite()':
/home/jearlam/.arduino15/packages/rp2040/hardware/rp2040/2.6.2/libraries/SDFS/src/SDFS.h:279:31: error: 'using element_type = class File32' {aka 'class File32'} has no member named 'availableSpaceForWrite'; did you mean 'availableForWrite'?
  279 |         return _opened ? _fd->availableSpaceForWrite() : 0;
      |                               ^~~~~~~~~~~~~~~~~~~~~~
      |                               availableForWrite
/home/jearlam/Arduino/libraries/CSV-Parser-for-Arduino/CSV_Parser.cpp: In constructor 'CSV_Parser::CSV_Parser(const char*, const char*, bool, char, char)':
/home/jearlam/Arduino/libraries/CSV-Parser-for-Arduino/CSV_Parser.cpp:72:14: warning: converting to non-pointer type 'int' from NULL [-Wconversion-null]
   72 |   rows_count(NULL),
      |              ^~~~
/home/jearlam/Arduino/libraries/CSV-Parser-for-Arduino/CSV_Parser.cpp:82:29: warning: list-initializer for non-class type must not be parenthesized
   82 |   header_parsed(!has_header_)
      |                             ^
/home/jearlam/Arduino/libraries/CSV-Parser-for-Arduino/CSV_Parser.cpp:81:15: warning: converting to non-pointer type 'uint16_t' {aka 'short unsigned int'} from NULL [-Wconversion-null]
   81 |   current_col(NULL),
      |               ^~~~
Multiple libraries were found for "SdFat.h"
  Used: /home/jearlam/Arduino/libraries/SdFat
  Not used: /home/jearlam/.arduino15/packages/rp2040/hardware/rp2040/2.6.2/libraries/ESP8266SdFat
Multiple libraries were found for "SD.h"
  Used: /home/jearlam/.arduino15/packages/rp2040/hardware/rp2040/2.6.2/libraries/SD
  Not used: /home/jearlam/.arduino15/libraries/SD
  Not used: /home/jearlam/Arduino/libraries/SD
exit status 1

Compilation error: exit status 1
```

Is this an issue with the library, or an incomplete/incompatible implementation of SD on the RP2040?

Thanks!

michalmonday, 2022-10-27:
Hello, I tried to reproduce the issue, I installed the 2.6.2 package with RP2040 boards:

[image]

I opened the basic_usage.ino example and selected Adafruit RP2040:

[image]

I tried to compile it and it finished successfully (I tried both: with and without CSV_PARSER_DONT_IMPORT_SD definition). My Arduino version is 1.8.19 (not sure if it makes much difference).

Are you using the same RP2040 boards package as I tried? 
Which board are you using?

JoeEarlam, 2022-10-27:
Hi, I'm using the same board package with a standard Pi Pico. I am however
using IDE 2.0.0.

michalmonday, 2022-10-27:
I just tried Arduino 2.0.1, using Pi Pico and it still compiled well, it shows some warnings too but completes in the end.

[image]

Could you check the version of CSV parser you have? (I have 1.0.0) 

Btw I think this part of output may be indicating the issue here:
> Multiple libraries were found for "SdFat.h"
  Used: /home/jearlam/Arduino/libraries/SdFat
  Not used: /home/jearlam/.arduino15/packages/rp2040/hardware/rp2040/2.6.2/libraries/ESP8266SdFat
Multiple libraries were found for "SD.h"
  Used: /home/jearlam/.arduino15/packages/rp2040/hardware/rp2040/2.6.2/libraries/SD
  Not used: /home/jearlam/.arduino15/libraries/SD
  Not used: /home/jearlam/Arduino/libraries/SD

michalmonday, 2022-10-27:
These are SD libraries displayed in my library manager after searching "sd" and selecting "data storage" topic:

[image]

[image]

[image]

I don't know how to fix this problem to be honest...

JoeEarlam, 2022-10-28:
I deleted the SDfat library and it now compiles correctly. Also, if I add #define CSV_PARSER_DONT_IMPORT_SD to the library's .cpp file it compiles correctly, even with the SDfat library installed. Doing some [reading](https://forum.arduino.cc/t/need-to-set-a-define-in-a-sketch-so-a-library-can-compile-correctly/140113) it turns out that Arduino IDE doesn't support this behaviour. 

Maybe the best thing to do here is to split the SD functionality into a separate header file, which can be included if required?