michalmonday/CSV-Parser-for-Arduino #21: Compile warnings in CSV_Parser.cpp

RealJamieRI, 2023-04-09:
Should I care about some compile warnings? 

I installed the CSV Parser 1.0.1 library in the Arduino IDE 2.0.4  and adding #include <CSV_Parser.h> to one of my source files. 

Without other code, the IDE compiler says: 

`c:\Users\jamie\OneDrive\Documents\Arduino\libraries\CSV_Parser\CSV_Parser.cpp: In constructor 'CSV_Parser::CSV_Parser(const char*, const char*, bool, char, char)':
c:\Users\jamie\OneDrive\Documents\Arduino\libraries\CSV_Parser\CSV_Parser.cpp:72:14: warning: converting to non-pointer type 'int' from NULL [-Wconversion-null]
   72 |   rows_count(NULL),
      |              ^~~~
c:\Users\jamie\OneDrive\Documents\Arduino\libraries\CSV_Parser\CSV_Parser.cpp:82:29: warning: list-initializer for non-class type must not be parenthesized
   82 |   header_parsed(!has_header_)
      |                             ^
c:\Users\jamie\OneDrive\Documents\Arduino\libraries\CSV_Parser\CSV_Parser.cpp:81:15: warning: converting to non-pointer type 'uint16_t' {aka 'short unsigned int'} from NULL [-Wconversion-null]
   81 |   current_col(NULL),
      |               ^~~~
`
TLDR

- I am using Arduino IDE 2.0.4. 
- building successfully so far Adafruit Feather M4 CAN (SAME51) plus an RTC / SD card FeatherWing (https://learn.adafruit.com/adafruit-adalogger-featherwing) 
- I want to store and retrieve thermistor calibration values for multiple thermistors. (in my evolving and functioning datalogger app)

About me:  I am a career EE just learning C++, Github, and Arduino with long past coding experience using K&R Microsoft C in the 90's ... This being about your library, I thought a library warning would be of interest here. Is that right?

Thanks! Jamie

Jusufs, 2023-10-19:
I have also some warnings. I tried open ai chat and make some hope good code corrections in CSV_Parser.cpp
`CSV_Parser::CSV_Parser(const char * s, const char * fmt_, bool has_header_, char delimiter_, char quote_char_) :
  fmt( strdup_ignoring_u(fmt_) ),
 //Jozef 19.10.2023
  //rows_count(NULL),
  rows_count(0),
//Jozef 
  cols_count( strlen_ignoring_u(fmt_) ),
  has_header(has_header_),
  delimiter(delimiter_),
  quote_char(quote_char_),
  delim_chars({'\r', '\n', delimiter_, 0}),
  //whole_csv_supplied(false),
  whole_csv_supplied(bool(s) ? true : false), // in constructor where whole csv is not supplied at once it should be set to false
//Jozef 19.10.2023  
  //leftover(NULL),
  //current_col(NULL),
  leftover(0),
  current_col(0), 
  
  header_parsed(!has_header_)
//  header_parsed = !has_header_;`

But still one warning:
`c:\Users\jozef\Documents\Arduino\libraries\CSV_Parser\CSV_Parser.cpp: In constructor 'CSV_Parser::CSV_Parser(const char*, const char*, bool, char, char)':
c:\Users\jozef\Documents\Arduino\libraries\CSV_Parser\CSV_Parser.cpp:89:29: warning: list-initializer for non-class type must not be parenthesized
   header_parsed(!has_header_)
                             ^`

michalmonday, 2023-10-19:
@RealJamieRI Hello, sorry for such late reply, I completely forgot about the issue.

@Jusufs Thank you very much for suggestions to fix it.

In the 1.2.1 version there should be no warnings.

Jusufs, 2023-10-20:
Thanks for rapid reaction Michal

Jusufs, 2023-10-21:
Please insert this information about decimal delimiter '.' in comments of CSV_Parser package.

closed 2023-10-21 by michalmonday