michalmonday/CSV-Parser-for-Arduino #25: 'Namespace' object has no attribute 'csv', reading_from_computer_python

nenp93, 2023-12-19:
Hello,
I'm completely new to arduino and python. I need to give my arduino due coordinates from a csv file, which a programs creates on my pc. The arduino is used to controll a robot arm. I'm trying to dive in arduino an python programming and tried to run the example ,,reading_from_computer_python''. I copied the sketch for arduino and uploaded it successfully. When I try to run python --csv hurricanes-csv in the command window on my pc, the following error occures: line 42, in <module>
    with open(args.csv .name, 'rb') as f:
              ^^^^^^^^
AttributeError: 'Namespace' object has no attribute 'csv'. Did you mean: 'csv '?

What am I doing wrong?

michalmonday, 2023-12-19:
Hello,
It looks like in your example theres a space between `args.csv` and `.name`. I think removing that space should fix the error, but let me me know if that's not the case.

It doesn't look like thats space is in the original [file](https://github.com/michalmonday/CSV-Parser-for-Arduino/blob/master/examples/reading_from_computer_python/arduino_serial.py):
[image]

michalmonday, 2023-12-19:
In Python strings need to be enclosed in single or double quote characters like:

```python
s.port = "COM3"
```

Otherwise the program would think that "COM3" is a variable. Btw if you take a look at the [arduino_serial.py](https://github.com/michalmonday/CSV-Parser-for-Arduino/blob/master/examples/reading_from_computer_python/arduino_serial.py) example, it automatically finds a port with "arduino" in its name, if multiple ports like that are found then it lets the user to select which one to use.

nenp93, 2023-12-20:
Allright it worked. But the next issue is right around the corner. Independant which USB port from my Arduino Due I am using, the command window doesn't receive or doesn't print the data. After I run the command, it looks like on the screenshot and I can only close the window. I can't type anything.
[image]

michalmonday, 2023-12-20:
Is Arduino Due running the [reading_from_computer_python.ino](https://github.com/michalmonday/CSV-Parser-for-Arduino/blob/master/examples/reading_from_computer_python/reading_from_computer_python.ino) example?

nenp93, 2023-12-20:
yes

michalmonday, 2023-12-20:
It looks like the [reading_from_computer_python.ino](https://github.com/michalmonday/CSV-Parser-for-Arduino/blob/master/examples/reading_from_computer_python/reading_from_computer_python.ino) example has a bug that only happens when clock of the board is too fast.

A quick workaround would be to replace this line:
```cpp
cp << Serial.read();
```

with this line:
```cpp
cp << Serial.readString();
```

I'm going to fix it properly in the future

nenp93, 2023-12-20:
unfortunately the quick workaround does'nt solve the problem. I will keep on trying.

michalmonday, 2023-12-26:
I modified the [reading_from_computer_python.ino](https://github.com/michalmonday/CSV-Parser-for-Arduino/blob/master/examples/reading_from_computer_python/reading_from_computer_python.ino) example and added the new one [reading_from_computer_python_row_by_row](https://github.com/michalmonday/CSV-Parser-for-Arduino/tree/master/examples/reading_from_computer_python_row_by_row) which should use much less memory.